sparkfun/SparkFun_CCS811_Arduino_Library #3: Unable to Initialize CCS811

cobracmder, 2017-05-09:
Howdy.

I'm unable to initialize the CCS811 module on my Qwiic combo board (CCS811 + BME280) on:
* ESP32 + Qwiic shield
* Arduino 101 + Qwiic shield
* Arduino Pro Mini + jumpers

The BME280 works just fine.

Here's the sample program (I've modified SparkFunCCS811.h and cpp to include to additional error conditions: BEGINCORE_ERROR and BEGIN_ERROR to determine begin function was failing):
```
#include <SparkFunCCS811.h>

CCS811 ccs811(0x5B);

void print_ccs811_begin_error(CCS811Core::status errorCode)
{
  switch (errorCode)
  {
  case CCS811Core::SENSOR_SUCCESS:
    Serial.println("SUCCESS");
    break;
  case CCS811Core::SENSOR_ID_ERROR:
    Serial.println("ID_ERROR");
    break;
  case CCS811Core::SENSOR_I2C_ERROR:
    Serial.println("I2C_ERROR");
    break;
  case CCS811Core::SENSOR_INTERNAL_ERROR:
    Serial.println("INTERNAL_ERROR");
    break;
  case CCS811Core::SENSOR_GENERIC_ERROR:
    Serial.println("GENERIC_ERROR");
    break;
  case CCS811Core::BEGINCORE_ERROR:
    Serial.println("BEGINCORE_ERROR");
    break;
  case CCS811Core::BEGIN_ERROR:
    Serial.println("BEGIN_ERROR");
    break;
  default:
    Serial.println("Unspecified error.");
  }
}

void print_ccs811_error()
{
  uint8_t error = ccs811.getErrorRegister();

  if (error == 0xFF) //comm error
  {
    Serial.println("Failed to get ERROR_ID register.");
  }
  else
  {
    Serial.print("Error: ");
    if (error & 1 << 5) Serial.print("HeaterSupply");
    if (error & 1 << 4) Serial.print("HeaterFault");
    if (error & 1 << 3) Serial.print("MaxResistance");
    if (error & 1 << 2) Serial.print("MeasModeInvalid");
    if (error & 1 << 1) Serial.print("ReadRegInvalid");
    if (error & 1 << 0) Serial.print("MsgInvalid");
    Serial.println();
  }
}

void setup() {
  Serial.begin(9600);
  CCS811Core::status rc;

  rc = ccs811.begin();
  if (rc != CCS811Core::SENSOR_SUCCESS)
  {
    print_ccs811_begin_error(rc);
  }
  else
  {
    Serial.println("CCS81 online");
  }
}

void loop() {
  if (ccs811.dataAvailable())
  {
    ccs811.readAlgorithmResults(); // this updates tVOC and eCO2 variables
  }
  else if (ccs811.checkForStatusError())
  {
    print_ccs811_error();
  } 
  delay(2000);
}
```

Here's where the library is failing: https://github.com/sparkfun/SparkFun_CCS811_Arduino_Library/blob/master/src/SparkFunCCS811.cpp#L80-L82 (returns I2C Error).

Here's the output from Logic monitoring the I2C bus:
```
Time [s], Analyzer Name, Decoded Protocol Result
0.138571250000000,I2C,Setup Write to [[ (0x5B)] + NAK
0.138716000000000,I2C,Setup Read to [[ (0x5B)] + NAK
0.139131750000000,I2C,Setup Write to [[ (0x5B)] + NAK
0.139276500000000,I2C,Setup Read to [[ (0x5B)] + NAK
0.139436500000000,I2C,Setup Write to [[ (0x5B)] + NAK
0.139581250000000,I2C,Setup Read to [[ (0x5B)] + NAK
2.148323500000000,I2C,Setup Write to [[ (0x5B)] + NAK
2.148468250000000,I2C,Setup Read to [[ (0x5B)] + NAK
2.148628000000000,I2C,Setup Write to [[ (0x5B)] + NAK
2.148773000000000,I2C,Setup Read to [[ (0x5B)] + NAK
4.157516750000000,I2C,Setup Write to [[ (0x5B)] + NAK
4.157661500000000,I2C,Setup Read to [[ (0x5B)] + NAK
4.157821250000000,I2C,Setup Write to [[ (0x5B)] + NAK
4.157966250000000,I2C,Setup Read to [[ (0x5B)] + NAK
6.166705000000000,I2C,Setup Write to [[ (0x5B)] + NAK
6.166850000000000,I2C,Setup Read to [[ (0x5B)] + NAK
6.167010000000000,I2C,Setup Write to [[ (0x5B)] + NAK
6.167154750000000,I2C,Setup Read to [[ (0x5B)] + NAK
8.175893750000000,I2C,Setup Write to [[ (0x5B)] + NAK
8.176038250000000,I2C,Setup Read to [[ (0x5B)] + NAK
8.176198250000001,I2C,Setup Write to [[ (0x5B)] + NAK
8.176342999999999,I2C,Setup Read to [[ (0x5B)] + NAK
```

Any pointers for additional debugging?

codmpm, 2017-05-21:
I think I have a very similar if not the same problem here with the CCS811 together with the BME280. I don't have the Qwiic Board but I wired the Sparkfun CCS811 together with an Adafruit BME280 in I2C mode.

I had it working with the [BME280Compensated](https://github.com/sparkfun/SparkFun_CCS811_Arduino_Library/tree/master/examples/BME280Compensated) example on an Arduino Uno 5V without any problems. I also got it working on a Freaduino Uno (ATMega 328p) which can run at 3.3V to ensure that this is not a voltage issue.

On an ESP12-F with current ESP Arduino 2.3.0 and a freshly cloned CCS811 library I get an`I2C_ERROR`.

I did not patch `SparkFunCCS811.h` as @cobracmder did.

liquidshock, 2017-08-19:
I had the same issue, but I managed to fix it by stretching the I2C clock in the  "core_esp8266_si2c.c" file. I have doubled it, and all the sudden all the devices on the bus with CCS811 started to work. So far, I haven't noticed any side effect of this change.

codmpm, 2017-08-20:
I will test this as soon as I can 👍

vibnwis, 2017-09-10:
"I had the same issue, but I managed to fix it by stretching the I2C clock in the "core_esp8266_si2c.c" file."
I am using esp32 WROOM. what file is equivalent to core_esp8266_si2c.c?

vibnwis, 2017-09-10:
Hm reduced the bus speed from 1MHz to 400KHz, it is working now.

maddhin, 2017-10-02:
I'm not very familiar with programming, could you explain what exactly should be changed in the "core_esp8266_si2c.c" file? Thanks in advance!

nseidle, 2018-10-12:
We've added an increase to the clock stretching to 200ms [here](https://github.com/sparkfun/SparkFun_CCS811_Arduino_Library/blob/master/src/SparkFunCCS811.cpp#L61). This should take care of the issue. If you continue to have problems, please let us know.

closed 2018-10-12 by nseidle

uvedhe, 2019-01-04:
I had the same problem, and using the SparkFun CCS811 library version 1.0.0 (instead of 1.0.1) fixed the issue!

uvedhe, 2019-02-19:
This issue persists in v1.0.7 - had to downgrade to 1.0.3 for it to work again! (1.0.0 wasn't available through the Library manager)

cjmssmd, 2019-03-18:
BOOM!  Thanks uvedhe.  I had the same problem and downgrading to 1.0.3 worked for me.  My config is a new redboard qwiic https://www.sparkfun.com/products/15123 and a new Combo Breakout - CCS811/BME280 https://www.sparkfun.com/products/14348.