sparkfun/SparkFun_CCS811_Arduino_Library #5: working with CJMCU-811 & Wemos D1 mini?

maddhin, 2017-10-02:
For me, being abroad, the Sparkfun board is not available but I got the CJMCU-811 board with the CCS811. I tried to get it to work but failed so far. I'm a noob and not sure what I am doing wrong. I think I have configured all PINs correct etc but still no result. The only thing I get is a: 

⸮CCS811 Basic Example
.begin() returned with an error.

Soft WDT reset

ctx: cont 
sp: 3ffef2f0 end: 3ffef4d0 offset: 01b0

>>>stack>>>
3ffef4a0:  3fffdad0 00000000 3ffee480 402020ac  
3ffef4b0:  feefeffe feefeffe 3ffee49c 40202cb4  
3ffef4c0:  feefeffe feefeffe 3ffee4b0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

Does anybody has a good idea on how to get this to work? It seems this is the only "large" CCS811 libary and it would be great to get this working!
Many thanks in advance!

maddhin, 2017-10-03:
I tried the WakeAndInterrupt.ino example and at least get a proper error now:
⸮
...
CCS811 begin exited with: I2C_ERROR
Mode request exited with: I2C_ERROR
Interrupt configuation exited with: I2C_ERROR

The problem seems to be that the I2C communication is not working when only SDA/SCL are connected. Does anybody has an idea how to get this going? I am certain I connected the CCS811 with the right SDA/SCL pins yet communication doesn't seem possible.

coolioxlr, 2017-10-06:
Getting the same error too

nseidle, 2018-10-12:
Hmm hard to say. Do you have pull up resistors on the SDA/SCL lines?

coelner, 2018-11-16:
@maddhin Decode the stack trace with this: https://github.com/me-no-dev/EspExceptionDecoder
The exception is raised due the While(1); statement

- Did you test another i2c hardware? This library restart the I2C bus with the default pins, which are maybe not the one used by you.
- Did you tried the other I2C address?

@nseidle Pullup resistors are embedded on that specific board.

bertrik, 2019-01-03:
It works for me with a Wemos D1 mini and the CJMCU-811 board. I use pin D2 going to WAK, D3 going to SDA, D4 going to SCL, with the sparkfun basic example sketch (Example1_BasicReadings.ino) modified for these pins. Pin D2 is configured as output and set low (0 volts), this is essential. My sketch uses I2C address 0x5A. My experience with the CJMCU-811 board is documented at https://revspace.nl/CJMCU-811

pluskal, 2019-01-19:
@maddhin, try the second address *0x5A*. Worked for me on Wemos D1 mini and the CJMCU-811.

[Example 1](https://github.com/sparkfun/SparkFun_CCS811_Arduino_Library/tree/master/examples/Example1_BasicReadings) - default pinout VCC +3.3, GND, SCL-D1, SDA-D2, WAK-GND, just change the address to *0x5A*.

enry86cami, 2021-11-10:
my CJMCU-811 works with an arduino nano but doesn't with a wemos D1 mini....
This is my wiring:
[image]

And this is the code:
`#include "Adafruit_CCS811.h"
Adafruit_CCS811 ccs;
void setup() {
Serial.begin(9600);
Serial.println("CCS811 test");
if(!ccs.begin()){
Serial.println("Failed to start sensor! Please check your wiring.");
while(1);
}
//calibrate temperature sensor
while(!ccs.available());
float temp = ccs.calculateTemperature();
ccs.setTempOffset(temp - 25.0);
}
void loop() {
if(ccs.available()){
float temp = ccs.calculateTemperature();
if(!ccs.readData()){
Serial.print("CO2: ");
Serial.print(ccs.geteCO2());
Serial.print("ppm, TVOC: ");
Serial.print(ccs.getTVOC());
Serial.print("ppb Temp:");
Serial.println(temp);
}
else{
Serial.println("ERROR!");
while(1);
}
}
delay(500);
}`