sparkfun/SparkFun_CCS811_Arduino_Library #6: I2C error on CCS811.begin()

cmsavage, 2017-12-01:
I'm working with the CCS811+BME280 combo board attached to an ESP8266 (NodeMCU).  The BME280 can be initialized and sensor values read over the I2C bus, but once I run CCS811.begin(), neither component seems to work properly over I2C.  I discovered that adding a brief delay between starting the app start and setting the drive mode in the begin() routine solves the problem.  Here is my own fix in SparkFunCCS811.cpp:
```
	//Write 0 bytes to this register to start app
	Wire.beginTransmission(I2CAddress);
	Wire.write(CSS811_APP_START);
	if( Wire.endTransmission() != 0 )
	{
		return SENSOR_I2C_ERROR;
	}	
	// Without a delay here, the CCS811 and I2C can be put in a bad state.
	// Seems to work with 50us delay, but make a bit longer to be sure.
	delayMicroseconds(100);
	
	returnError = setDriveMode(1); //Read every second
```

BriscoeTech, 2018-06-08:
Thank you so much for posting this, its exactly what I needed. This problem exists in all esp8266 from what I can tell, I tried it with a Sparkfun esp8266 Dev and a Adafruit esp8266 Huzzah and they were having the same issue. Copied in your code, works like a charm.

Sparkfun, Id definitely merge this change into your code.

Thanks again!

ParthSareen, 2018-08-09:
This was a great fix! I

mouse256, 2018-08-28:
Have been looking for a couple of hours why it was not working, this fix solves it. They should really take it in

nseidle, 2018-10-12:
While I appreciate the issue (and the PR! - thank you!) I'm very skeptical of adding delays just 'because they work'. I've added your recommendation with compiler guards for the ESP arch. If anyone finds a reason why this works I'd love to hear it.

closed 2018-10-12 by nseidle