sparkfun/SparkFun_CCS811_Arduino_Library #11: ESP8266 error

hsadeghilam, 2018-09-26:
I just updated my library and got an error:

C:\Users\XXXX\Documents\Arduino\libraries\SparkFun_CCS811_Breakout\src\SparkFunCCS811.cpp:61:8: error: macro names must be identifiers

 #ifdef (ARDUINO_ARCH_ESP32 ARDUINO_ARCH_ESP8266)

per1234, 2018-09-27:
Duplicate of https://github.com/sparkfun/SparkFun_CCS811_Arduino_Library/issues/9, which was already fixed by https://github.com/sparkfun/SparkFun_CCS811_Arduino_Library/pull/10.

There hasn't been a release since the fix so you would either need to download and install the beta version of the library from GitHub or just use Library Manager to roll back to the last good release.

nseidle, 2018-10-12:
Sorry. This will be fixed momentarily.

closed 2018-10-12 by nseidle