sparkfun/SparkFun_CCS811_Arduino_Library #12: SparkFun ESP32 Thing compile error

cap9qd, 2018-10-28:
I have an issue compiling this library when using the latest ESP32 Arduino Core and Arduino 1.8.5.  

I followed the [SparkFun ESP32 Thing hookup guide](https://learn.sparkfun.com/tutorials/esp32-thing-hookup-guide#installing-the-esp32-arduino-core) to setup the environment and tested it on a ESP32-BIT breakout from Amazon before receiving the ESP32 THING in the mail with the environmental monitor shield.

I get an error: 
`SparkFunCCS811.cpp:62:7:` error: 'class TwoWire' has no member named 'setClockStretchLimit'"`

I am using the example sketch for the Environmental Monitor Shield [setup guide](https://learn.sparkfun.com/tutorials/esp32-environment-sensor-shield-hookup-guide#software) and the libraries as downloaded by the Ardiuno library manager.  I did try to pull the latest commit of this library since in other issues (#11 ESP8266 error) it was mentioned that one issue I had was already fixed. 

Any ideas what I am doing wrong?  I tried pulling the latest the ESP32 stuff but this didnt fix it.  However, this is only the second project I have done with ESP32 so maybe I did something wrong updating the toolchain/core.

I commented out the WU stuff since I plan to push the data to a mosquito server instead.
```
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>

#include <SparkFunCCS811.h>
#include "SparkFunBME280.h"
#include "Wire.h"
#include <Sparkfun_APDS9301_Library.h>

BME280 bme;
CCS811 ccs(0x5B);
APDS9301 apds;

// Variables for wifi server setup 
char* ssid     = "";
char* password = ""; 
//String ID = "station_id_goes_here";
//String key = "wunderground_key_goes_here";  
WiFiClient client;
//const int httpPort = 80;
//const char* host = "weatherstation.wunderground.com";

// Variables and constants used in calculating the windspeed.
volatile unsigned long timeSinceLastTick = 0;
volatile unsigned long lastTick = 0;

// Variables and constants used in tracking rainfall
#define S_IN_DAY   86400
#define S_IN_HR     3600
#define NO_RAIN_SAMPLES 2000
volatile long rainTickList[NO_RAIN_SAMPLES];
volatile int rainTickIndex = 0;
volatile int rainTicks = 0;
int rainLastDay = 0;
int rainLastHour = 0;
int rainLastHourStart = 0;
int rainLastDayStart = 0;
long secsClock = 0;

String windDir = "";
float windSpeed = 0.0;

// Pin assignment definitions
#define WIND_SPD_PIN 14
#define RAIN_PIN     25
#define WIND_DIR_PIN 35
#define AIR_RST      4
#define AIR_WAKE     15
#define DONE_LED     5

void setup() 
{
  delay(5);    // The CCS811 wants a brief delay after startup.
  Serial.begin(115200);
  Wire.begin();

[... 258 lines cut ...]
}

// For the purposes of this calculation, 0deg is when the wind vane
//  is pointed at the anemometer. The angle increases in a clockwise
//  manner from there.
void windDirCalc(int vin)
{
  if      (vin < 150) windDir="202.5";
  else if (vin < 300) windDir = "180";
  else if (vin < 400) windDir = "247.5";
  else if (vin < 600) windDir = "225";
  else if (vin < 900) windDir = "292.5";
  else if (vin < 1100) windDir = "270";
  else if (vin < 1500) windDir = "112.5";
  else if (vin < 1700) windDir = "135";
  else if (vin < 2250) windDir = "337.5";
  else if (vin < 2350) windDir = "315";
  else if (vin < 2700) windDir = "67.5";
  else if (vin < 3000) windDir = "90";
  else if (vin < 3200) windDir = "22.5";
  else if (vin < 3400) windDir = "45";
  else if (vin < 4000) windDir = "0";
  else windDir = "0";
}
```

nseidle, 2018-11-04:
Sorry. Looks like we broke it a [few commits back](https://github.com/sparkfun/SparkFun_CCS811_Arduino_Library/commit/f2419e5b986a4a0882c02920797047f537986f74) when we were fixing clock stretching for ESP8266.

I don't have a good setup to test but it does now compile. I've pushed a release. Please grab v1.0.4 of the library and kick the tires. Let me know if you still have issues.

closed 2018-11-04 by nseidle

cap9qd, 2018-11-08:
Thanks Nate! I tested it on the ESP32 THING and and weather shield and it looks good!

nseidle, 2018-11-08:
Yay! Sorry about that. Thanks for confirming!