sparkfun/SparkFun_CCS811_Arduino_Library #14: Error compiling with library installed from library manager

wero1414, 2019-01-16:
I can see that the problem was fixed by this PR #10, so i think that the problem is that you havent done realese of this update in order to update the library on the arduino library manager

nseidle, 2019-01-24:
Sorry folks! I broke it. I just pushed v1.0.7 to try to fix this. 

There's something fishy going on with the Arduino library manager as well

[image]

The library isn't showing up at all. I'll check it tomorrow and file an issue with Arduino if it doesn't resolve over night.

bboyho, 2019-06-01:
I was updating something else with the CCS811 when I came across this issue. Testing v1.0.7 out, I was able to compile, upload the code, and read the sensor.

It looks like the issue is fixed so I will be closing this issue out. If we need  we can always open this issue back up. ;)

closed 2019-06-01 by bboyho