sparkfun/SparkFun_CCS811_Arduino_Library #15: Unable to use with software i2c

supersjimmie, 2019-03-18:
My sensors are connected to an arduino pro mini.
I use Testato software i2c which works fine with the bme280 and an ssd1306 display.
(bme280 with the known modifications in the SparkFun bme280 library)

This sparkfun ccs811 library works fine with the normal Wire i2c but when I try to use the above software i2c the begin-function just doesn't seem to return at all.

The base is: 
```
#include <SoftwareWire.h>
SoftwareWire myWire( A2, A3, true, true); // sdaPin, sclPin, pullups, detectClockStretch

#include "SparkFunCCS811.h"
#define CCS811_ADDR 0x5A //Alternate I2C Address
CCS811 mySensor(CCS811_ADDR);
```

The original i2c is on A4, A5 (currently unused to isolate all other possible issues), the software i2c is on A2, A3 (as shown).

In my setup() I have:
```
  Serial.println("Init myWire.");
  myWire.begin();
```

```
  Serial.println("init ccs811");
  CCS811Core::status returnCode = mySensor.begin(myWire);
  if (returnCode != CCS811Core::SENSOR_SUCCESS) {
    Serial.println("ccs811 .begin() returned with an error.");
  }

  delay(500);
  Serial.println("Start.");
```
Which prints the init mywire and the init ccs811 but then no error and no "Start.". It just seems to stay in the mySensor.begin(myWire);.

Also this same software i2c setup works fine with e.g. the ssd1306 display and the bme280.
So there's no issue with the software i2c and also not with the ccs811 hardware (because it does work when I revert to the basic hardware i2c wire on A4/A5).

nseidle, 2019-03-18:
> (bme280 with the known modifications in the SparkFun bme280 library)

Tell me more.

I'm guessing the problem is that this lib is expecting a TwoWire type pointer:

[image]

And I'm not sure what SoftwareWire is passing when you call a 

    mySensor.begin(myWire);

Additionally, what bothers me is that this code compiles at all. But the I2C comm fails immediately at the first attempt to write to a reg

[image]

I believe to make this lib work with software wire add a begin with SoftwareWire type def

https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/master/src/SparkFunBME280.h#L184

as well as a series of switch gates anywhere the I2C communication is made:

https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/master/src/SparkFunBME280.cpp#L614

It's big and ugly. Ideally software libs would init a full TwoWire type software interface. I'm not sure why they don't...

supersjimmie, 2019-03-19:
Yes it's easy to say that the problem is at the software wire, but when I change the ccs811 library by ONLY changing the ```_i2cport->```  to ```myWire.``` (and include the softwarewire lib), it works fine.
So the ccs811 and the softwarewire CAN work together, just not the way that the ccs811 library takes the softwarewire as a parameter.

nseidle, 2019-03-19:
If you go through the lib and change the arrow operators to dot operators, sure, it will work but you won't be able to pass in the wire port by pointer.

https://www.tutorialspoint.com/cplusplus/cpp_member_operators.htm

If you think you've got a workable solution I would be thrilled to review a PR.

supersjimmie, 2019-03-20:
So what you are saying this sparkfun thing won't work with the softwarewire libs that work for the sparkfun bme280. Can you advice what other software i2c would work then? (Preferably for both this ccs811 as well as for the bme280?)
Has this ccs811 way of accepting any other wire than the default been tested with any other?

nseidle, 2019-03-28:
Hi - This library is written to work with hardware I2C including multiple ports (Wire1, etc). We currently don't support software I2C.

closed 2019-03-28 by nseidle