sparkfun/SparkFun_CCS811_Arduino_Library #18: readNTC()  error

CostInCode, 2019-08-01:
I'm unable to read resistance (always 0 ohms) and temperature (always -273.15 deg) values. 
Whenever myCCS811.readNTC() is called I get the error ReadRegInvalid from printSensorError(). The source code is the Example3_Thermistor_Compensation. The other values (CO2 and TVOC) are fine.

Setup: Sparkfun SAMD21 Dev Breakout, Sparkfun CCS811, SEN-00250 thermistor between NTC terminals.

Output from Example3_ThermistorCompensation: [https://i.imgur.com/lDaxZ11.jpg](url)

SampleInaJar, 2019-10-19:
I’m running into the same issue. Did you ever figure this out?

CostInCode, 2019-10-30:
Unfortunately no, I didn't. Ended up using a temperature sensor.

AndyEngland521, 2020-04-06:
Hi guys, it looks like this feature has actually been deprecated and is no longer included in the CCS811.

I'd recommend using another temperature sensor like the [SparkX BME680](https://www.sparkfun.com/products/14570) to do temperature compensation.

closed 2020-04-06 by AndyEngland521